RoyBerntsenDesign/V-King-CoreXY #3: Wrong bolt size on Y carrier to XY idler ASY in drawings and in BoM

martinroger, 2019-02-20:
The bolts in the drawing "Y-slider drawing" #12 that join the Y carriers to their respective XY idlers is indicated as M4x10, and in the most recent online assembly they are actually M5x14. The BoM does not list any M5x14, but does mention some M5x12. Unfortunately the M5x12 can barely catch one thread of the T-nut on the other side.

The correct size should be M5x14, Qty 4 minimum in the BOM. M5x16 might also work, but it is unverified.

[image]

RoyBerntsenDesign, 2019-02-22:
Thanks for pointing this out. The brackets was redesignen and I never got
to change the bom... I will do  that this weekend

Best regards/Vennlig hilsen
Roy Berntsen
M.Sc. Design Engineer
Tel: +47 928 26 012
E-mail: roy@pro3d.no
Web: www.pro3d.no

RoyBerntsenDesign, 2019-03-01:
Updated the BOM with M5x14

closed 2019-03-01 by RoyBerntsenDesign