RoyBerntsenDesign/V-King-CoreXY #7: V-King Belted Z axis not moving up uniformly

stoky996, 2020-01-01:
after assembling my v-king i noticed that when testing my z axis it was not moving up correctly causing bed leveling to fail when the bed moved upwards the right side moved up faster then the left causing it to become slanted and when moving downwards the left hand side slants tightening the left causes the right side to resume it's behavior i will upload a video tomorrow i should say that the other side is fine and does not move. i will upload a video tomorrow got other stuff to do and will be at work tomorrow.

RoyBerntsenDesign, 2020-01-01:
Thank you for the feedback. Can you go to discord or messenger support chat
to solve this is more easy to help you. Also the Facebook group can help

Messenger:
Discord: https://discord.gg/MRgjAqT
FB: https://www.facebook.com/groups/viking3d/

Best regards/Vennlig hilsen
Roy Berntsen
M.Sc. Design Engineer
Tel: +47 928 26 012
E-mail: roy@pro3d.no
Web: www.pro3d.no

stoky996, 2020-01-01:
Cheers I have joined the Facebook group and continuing it from there thanks for responding anyhow.

Regards jesse

closed 2020-01-01 by stoky996