RoyBerntsenDesign/V-King-CoreXY #10: M5 80D copper washer in assembly drawings

cscott, 2020-04-29:
This is presumably the "M5 GT2 Idler Shim" from the BOM?  What are the source and/or dimensions of this?

cscott, 2020-04-29:
In the "Base Mechanical Kit" from makersupplies.sk this seems to be what is listed as "M5 Mini Precision Shim 8x5x1 (25)".  So something like https://www.aliexpress.com/item/32812003364.html ?

RoyBerntsenDesign, 2020-04-29:
The best option shim is 1mm thick 5 bore and 6 or 7mm outer diameter. The
stock option has been 5x8x1 (boreXodXthick) and these should be ok to use
as well but someone noted the outer dia could be little big for some
bearings

cscott, 2020-04-29:
It seems that M5x7x0.5 shims are what's readily available on aliexpress, etc.  Presumably two of these could be used instead of the 5x8x1.

RoyBerntsenDesign, 2020-04-30:
Yes those are perfect