RoyBerntsenDesign/V-King-CoreXY #9: GT belt width in BOM

cscott, 2020-04-29:
The BOM doesn't say the width or length of the GT belt, and at the links provided both the belt and the idler and motor pulleys are available in a variety of widths.  In the assembly drawing for the "XY Belt Tension Drawing" it says "Use 3mm, 4mm, or 5mm smooth idlers here" -- is that the bore?  I'd assume you'd need to use 5mm to match the M5 bolts.

I'm guessing "20tooth width 6mm bore 5mm without teeth" for the smooth pulley, and "20tooth width 6mm bore 5mm (with teeth)" for the toothed pulley.  For the motor pulley, is the 20 tooth one with a 7mm width for the belt, 15mm overall height, and 5mm bore the correct one?

RoyBerntsenDesign, 2020-04-29:
Hi and thanks for the feedback. This is important info and I will update
the bom soon. I have 20t pulleys and idlers. All 5mm bore yes. Belt is 6mm
wide and I dont remember the length. Let me check if I have this noted

cscott, 2020-04-29:
By my math, the length of the each XY belt should be the same as the circumference of the XY frame.  It would be interesting to note how much "extra" was needed for securing it to the X slider, so that folks building different sizes could get a rough idea of how much extra to leave.

RoyBerntsenDesign, 2020-04-30:
So i just measured and get about 180 cm and suggest 200cm for each xy belt.
I get 150 cm for each z drive. This means 700cm as a minimum in total